RROrg/rr #8175: 你好，我想知道镜像中某个文件怎么生成的

wpp5632447, 2025-04-24:
您好，
我发现镜像中有
/mnt/p3/cks/bzImage-epyc7002-7.2-5.10.55.gz
我想知道，这个文件是怎么制作的
我想要打开内核的某个功能，希望生成配套的bzImage-epyc7002-7.2-5.10.55.gz，谢谢
我下载了
epyc7002-gcc1220_glibc236_x86_64-GPL.txz
和sa6400的linux-5.10.x.txz，在sa6400的docker spksrc中和centos（内核3.10）均无法编译，希望楼主能提供教程，谢谢

wpp5632447, 2025-04-24:
env信息如下
root@spksrca:/spksrc/dsm5# env
HOSTNAME=spksrca
PWD=/spksrc/dsm5
HOME=/root
LANG=C.UTF-8
TERM=xterm
ARCH=x86_64
SHLVL=1
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
CROSS_COMPILE=/spksrc/dsm5/x86_64-pc-linux-gnu/bin/x86_64-pc-linux-gnu-
OLDPWD=/spksrc/dsm5/linux-5.10.x
_=/usr/bin/env

RR:rr-25.4.0.img
addons:all
modules:all
lkms:

DSM:

model:sa6400
version:7.2.2最新

日志如下
root@spksrca:/spksrc/dsm5/linux-5.10.x# 
root@spksrca:/spksrc/dsm5/linux-5.10.x# make -j4
  DESCEND  objtool
  DESCEND  bpf/resolve_btfids
  CALL    scripts/atomic/check-atomics.sh
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  GEN     .version
  CHK     include/generated/compile.h
  UPD     include/generated/compile.h
  CC      init/version.o
  AR      init/built-in.a
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.btf
  BTF     .btf.vmlinux.bin.o
die__process: DW_TAG_compile_unit, DW_TAG_type_unit, DW_TAG_partial_unit or DW_TAG_skeleton_unit expected got INVALID (0x0) @ 1dcf3ba!
error decoding cu 
pahole: .tmp_vmlinux.btf: Invalid argument
  LD      .tmp_vmlinux.kallsyms1
  KSYMS   .tmp_vmlinux.kallsyms1.S
  AS      .tmp_vmlinux.kallsyms1.S
  LD      .tmp_vmlinux.kallsyms2
  KSYMS   .tmp_vmlinux.kallsyms2.S
  AS      .tmp_vmlinux.kallsyms2.S
  LD      vmlinux
  BTFIDS  vmlinux
FAILED: load BTF from vmlinux: Unknown error -2make: *** [Makefile:1179: vmlinux] Error 255

我发现镜像中有
/mnt/p3/cks/bzImage-epyc7002-7.2-5.10.55.gz
我想知道，这个文件是怎么制作的
我想要打开内核的某个功能，希望生成配套的bzImage-epyc7002-7.2-5.10.55.gz，谢谢
我下载了
epyc7002-gcc1220_glibc236_x86_64-GPL.txz
和sa6400的linux-5.10.x.txz，在sa6400的docker spksrc中和centos（内核3.10）均无法编译，希望楼主能提供教程，谢谢

wjz304, 2025-04-24:
https://github.com/RROrg/linux_dsm_epyc7002

wpp5632447, 2025-04-25:
感谢大神的仓库资源，目前成功修改参数并编译成功，我的步骤如下
git clone 仓库
cp synology/synoconfigs/epyc7002
export ARCH=x86_64
然后make menuconfig,选择network中的ip multicast和下属的pim v2，其他任何东西都没有动过

CONFIG_IP_MROUTE_COMMON=y
CONFIG_IP_MROUTE=y

CONFIG_IP_PIMSM_V2=y

epyc7002
make oldconfig && make prepare && make modules_prepare && make scripts
最后make出来的镜像替换原有镜像，没有替换initrd-dsm(因为我水平还不行)
但是启动后发现有很多这样的报错
25T15:18:12+08:00 nasss kernel: [  454.372011] BUG: workqueue lockup - pool cpus=24 node=0 flags=0x4 nice=0 stuck for xxxxs! 
关机会卡住(web和xshell都退出)，但是显示其终端还能操作，我怀疑与这个有关联

环境VMware测试平台
自编译内核
sa6400
RR v25.4.1 / DSM 7.2.2-72806 Update 3
替换回来原有的zImage-rr，问题就消失。下面附件为几个为故障时候的日志信息，如果尊敬的大神得空，请帮忙看看，我网上搜索了很多信息，没有得到有效的解決办法

[kern.txt](https://github.com/user-attachments/files/19905256/kern.txt)
[message.txt](https://github.com/user-attachments/files/19905258/message.txt)
[dmesg.txt](https://github.com/user-attachments/files/19905257/dmesg.txt)

wjz304, 2025-04-28:
https://github.com/RROrg/rr/releases/tag/25.4.2

wpp5632447, 2025-04-28:
谢谢大神

wpp5632447, 2025-04-28:
> https://github.com/RROrg/rr/releases/tag/25.4.2

谢谢大神

wpp5632447, 2025-04-28:
谢谢大神，您新编译的是可用的，我编写的组播所关联的代码能跑起来，非常感谢

但是我有个问题，为何我编译的内核，替换内核后，没有替换initrd-dsm，会有问题，如果大神得空，请帮忙指点下
我的操作步骤很简单，docker中下拉ubunt，然后docker中
git clone https://github.com/RROrg/linux_dsm_epyc7002.git
apt安装必要的编译环境
apt install -y curl build-essential bison flex dwarves ruby libncurses-dev libssl-dev libelf-dev git等
然后
wget https://cndl.synology.cn/download/ToolChain/toolchain/7.2-63134/AMD%20x86%20Linux%20Linux%205.10.55%20%28epyc7002%29/epyc7002-gcc1220_glibc236_x86_64-GPL.txz
tar -Jxf epyc7002-gcc1220_glibc236_x86_64-GPL.txz
export ARCH=x86_64
export CROSS_COMPILE=/dsm2/x86_64-pc-linux-gnu/bin/x86_64-pc-linux-gnu-
cd linux_dsm_epyc7002
cp synology/synoconfigs/epyc7002 .config
make oldconfig && make prepare && make modules_prepare && make scripts
然后检查include/generated/utsrelease.h是否有加号结尾即（#define UTS_RELEASE "5.10.55+"
）
最后直接make -j12,然后拷贝arch/x86/boot/bzimage替换u盘中/mnt/p3/zImage-dsm后，直接启动或通过rr启动
最后能够ssh进去，但是有问题，
在这，请问您，上述哪个流程上有问题吗？请指点下，谢谢。
运行日志如下，如果重装，就会出现上述前面提到的kern.log的日志，

18:18:09+08:00之前是我自己编译的内核跑出的日志
后面是更换为您的内核
全程都是用的老版本的initrd-dsm(估计是前几个版本u盘menu.sh生成的)

[messages.txt](https://github.com/user-attachments/files/19939137/messages.txt)

wjz304, 2025-04-28:
一两句说不清楚
QQ： 304403268 / Email：wjz304@qq.com

closed 2025-04-28 by wjz304