RROrg/rr #8183: SA6400纯nvme固态安装后重启丢失

wiseman815, 2025-04-25:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
型号:     SA6400(epyc7002)
版本:     7.2(72806)
内核:     custom
LKM:      prod
MEV:      physical
DMI:      GMKtec NucBox G3/GMKtec, BIOS GMK_G3 10/21/2023
CPU:      Intel(R) N100
MEM:      15743 MB

RR version: (You can find it in the update menu.)  
 * RR:  25.4.1
 * addons: hdddb、nvmecache、nvmesystem、nvmevolume
 * modules:  
 * lkms:  

DSM: 
 * model:  sa6400
 * version:  7.2-72806

Issue:  安装dsm后重启，ip一直不上线，已设置了静态ip

logs:  

按提示装了log插件，进去还是提示无log文件

wiseman815, 2025-04-25:
补充log

[logs.tar_2.gz](https://github.com/user-attachments/files/19902409/logs.tar_2.gz)

wiseman815, 2025-04-25:
#4587 我曾经遇到同样问题，当时未及时下载反馈的测试链接，未能进一步测试

wjz304, 2025-05-08:
日志没有异常，也没有 DHCP，目前无法调查

wjz304, 2025-05-08:
> [#4587](https://github.com/RROrg/rr/issues/4587) 我曾经遇到同样问题，当时未及时下载反馈的测试链接，未能进一步测试

新版本即可

Liang-Dai, 2025-06-12:
我是用最新出的版本V25.6.5也有同样的问题，可以获取IP，可以ping的通，但是无法通过5000访问，端口扫描可以看到只监听了80端口，访问80会自动跳转到5000，然后显示无法访问
RR:V25.6.5
型号：DS918+
版本：7.2(72806)