RROrg/rr #8184: BUG  esxi8.0虚拟化安装的RR 25.4.0   SA6400  默认的vmtools服务类型

lm010692, 2025-04-25:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:   25.4.0
 * addons:  vm-tools
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2.2

Issue:  
默认的vmtools服务类型为simple，与vmtools的启动方式不匹配会导致vm-tools服务无法正常启动，调整为forking后，正常运行，可以调整一下。

附带调整后的文件：
[Unit]
Description=RR addon vmtools daemon
IgnoreOnIsolate=true
After=multi-user.target

[Service]
**Type=forking
PIDFile=/var/run/vmtools.pid**
Environment="PATH=/usr/vmtools/bin:/usr/vmtools/sbin:$PATH"
Environment="LD_LIBRARY_PATH=/usr/vmtools/lib:$LD_LIBRARY_PATH"
ExecStart=/usr/vmtools/bin/vmtoolsd -c /usr/vmtools/etc/vmware-tools/tools.conf --common-path=/usr/vmtools/lib/open-vm-tools/plugins --plugi
n-path=/usr/vmtools/lib/open-vm-tools/plugins/vmsvc -b /var/run/vmtools.pid
Restart=always
RestartSec=10

[Install]
WantedBy=multi-user.target

仅仅添加了加粗的部分
最后感谢大大开发的RR，确实好用。

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

lm010692, 2025-04-25:
补充一个问题，开启自带的vm-tools会出现关机时触发shutdown失败，esxi的web页面里 手动点击虚拟机的shutdown按钮也是同样的情况，第一次点击会报错，再点击第二次就可以成功触发关机。 详情见下图。

[image]

wjz304, 2025-04-28:
https://github.com/RROrg/rr/releases/tag/25.4.2

lm010692, 2025-04-28:
测试最新版本25.4.2  关机报错提示变了，详情如下图所示：

[image]

[image]

wjz304, 2025-04-29:
我没有 esxi 环境, 晚点用vmware 试下吧

lm010692, 2025-04-29:
> 我没有 esxi 环境, 晚点用vmware 试下吧

理解理解，大佬  如有需要 我们可以约个时间 我配合测试，我单独整个测试版的syno

wjz304, 2025-05-07:
test 25.5.0

lm010692, 2025-05-08:
替换25.5.2的启动盘后，无法启动   报错如图所示。

[image]

wjz304, 2025-05-08:
> 替换25.5.2的启动盘后，无法启动 报错如图所示。
> 
> [image]

你测试这个干什么？

lm010692, 2025-05-08:
> > 替换25.5.2的启动盘后，无法启动 报错如图所示。
> > [image]
> 
> 你测试这个干什么？

之前是直接替换升级的，这次也就直接替换升级了，没想到翻车了，刚刚测试了在线升级是可以的，关机还是会报错。
[image]

发现个新问题：新建了一个虚拟机，用的25.5.2引导，发现无法在线选择型号了，只能选择解析pat，出去的网络肯定是没问题，有点奇怪。

wjz304, 2025-05-08:
open vm tools  全是硬编码，即使设置了 bin 和 lib 的路径 还是很多东西会找不到，😔

wjz304, 2025-05-08:
在 /var/log/ 下 有 4个 vmtools 的日志，你可以看看，

lm010692, 2025-05-08:
计划用journalctl -f 看看关机实时报错，搞了两回，一样的情况，都是直接崩了。  - -！ 这个和systemd服务无法获取系统shell的变量有关系吗。。。

wjz304, 2025-05-08:
QQ 或者TG 吧

lm010692, 2025-05-08:
> QQ 或者TG 吧

阔以呀   TG：sky

wjz304, 2025-05-08:
https://t.me/RR_Org

wjz304, 2025-05-15:
https://github.com/RROrg/rr/releases/tag/25.5.6

closed 2025-05-19 by lm010692