RROrg/rr #8349: DS925+

wjz304, 2025-04-27:


wjz304, 2025-04-27:
[image]

wjz304, 2025-04-27:
[image]

007revad, 2025-04-29:
There is also another way:

1. Open telnet via http://<NAS-IP>:5000/webman/start_telnet.cgi
2. Log in to telnet via root, and execute the following command,
    `touch /tmp/installable_check_pass`
3. Refresh the web installation page.

chnjinzhe, 2025-05-01:
目前似乎有pci卡就无法启动，另一台无pci网卡则顺利安装启动。

TheWorldIsGamble, 2025-05-05:
925不就已经不认其他硬盘了么？

PeterSuh-Q3, 2025-05-06:
> 925不就已经不认其他硬盘了么？

是的，正版DS925+已经被封了。

apriliars3, 2025-05-10:
I am testing DS925+ on Proxmox with Intel 8500 and not works. Same issue with DS1525+. Thanks

[image]

[image]