RROrg/rr #8602: PVE虚拟化RTL8125直通时RR shell/menu.sh下令poweroff无法关机

ctl26481, 2025-05-02:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU Standard PC (Q35 + ICH9, 2009), BIOS 3.20230228-4 06/06/2023
 * CPU:  Intel Core i5-7500
 * NIC:  M.2 A+E key转RTL8125BG，不带ROM款式[10ec:8125]

RR version: (You can find it in the update menu.)  
 * RR:  test 23.10.5 OK/24.2.0 4G OK/24.3.9 OK/24.4.0 OK/24.4.1 OK/24.4.2 NG/24.4.5 NG/24.4.8 NG/24.5.0 NG/24.7.2 NG/24.11.1 NG/25.4.2 NG
 * addons:  default/auto
 * modules:  default/auto
 * lkms:  prod

DSM: 
 * model:  仅测试DS3622xs+
 * version:  "7.2(72806)"

Issue:
PVE虚拟化RTL8125直通时RR shell/menu.sh下令poweroff无法关机

使用M.2 A+E key转RTL8125BG，不带ROM款式连接1Gbps交換機，无法在RR shell环境下达poweroff命令
PVE GUI左侧显示感叹号(internal-error)，必须在PVE选择VM后按右上角Stop强制关闭VM
直通Windows 11 VM无关机问题

目前回推RR版本测试24.4.1为最后可在RR shell/menu.sh下达poweroff的版本，24.4.2以上则NG(no-go)
上面RR version那串是已测试过的

若在无法shell/menu.sh下令poweroff的RR版本(RR≥24.4.2，包含最新版25.4.2):
 * 所有环境下正常DHCP
 * 使用modprobe -r r8125移除驱动后可从RR shell下达poweroff指令
(init.sh显示Network devices not found! Please re execute init.sh after connecting to the network!)
 * 在DS3622xs+/7.2/junior(欢迎页面)下可从串口(serial0)下达poweroff指令
 * DS环境可用GUI关机选项，无异常关机log
 * DS环境1Gbps正常传输

若在可从shell/menu.sh下令poweroff的RR版本(RR≤24.4.1):
 * 所有环境下正常DHCP
 * 所有环境下都可以正常poweroff，DS GUI关机选项无异常关机log
 * DS环境1Gbps正常传输

为了方便debug loader，手动在VMID.conf用模拟USB的方式:
`args: -device 'nec-usb-xhci,id=usb-bus0,multifunction=on' -drive 'file=/XXXXXX/rr.img,media=disk,format=raw,if=none,id=drive-disk-bootloader' -device 'usb-storage,bus=usb-bus0.0,port=1,drive=drive-disk-bootloader,id=usb-disk-bootloader,bootindex=999,removable=on'`

VMID.conf
> agent: 0
> balloon: 0
> bios: ovmf
> boot:
> cores: 4
> cpu: host
> hostpci0: 0000:05:00,pcie=1 **#RTL8125BG/All_Function/ROM-BAR/PCI-E**
> machine: q35
> memory: 4096
> meta: creation-qemu=9.2.0,ctime=XXXX
> name: 8125test
> numa: 0
> ostype: l26
> scsi0: VM:101/vm-101-disk-0.qcow2,backup=0,discard=on,iothread=1,size=64G,ssd=1
> scsihw: virtio-scsi-single
> serial0: socket
> smbios1: uuid=XXXX
> sockets: 1
> tablet: 0
> vmgenid: XXXX

logs:  
PVE syslog(25.4.2 RR shell无法poweroff，PVE显示internal-error时):
> May 02 20:12:02 pve QEMU[1563981]: error: kvm run failed Bad address
> May 02 20:12:02 pve QEMU[1563981]: RAX=ffff93ea405c0000 RBX=ffff8aab0b000000 RCX=0000000000000001 RDX=ffff8aab0b47e12c
> May 02 20:12:02 pve QEMU[1563981]: RSI=0000000135eb6000 RDI=ffff8aab0b000940 RBP=ffff8aab0b000940 RSP=ffff93ea40293bd0
> May 02 20:12:02 pve QEMU[1563981]: R8 =000000000000081d R9 =000000000001c10f R10=00000000000006da R11=0000000000000001
> May 02 20:12:02 pve QEMU[1563981]: R12=ffff8aaaf5eb6000 R13=0000000135eb6000 R14=0000000000000000 R15=0000000000000000
> May 02 20:12:02 pve QEMU[1563981]: RIP=ffffffffc0408013 RFL=00010202 [-------] CPL=0 II=0 A20=1 SMM=0 HLT=0
> May 02 20:12:02 pve QEMU[1563981]: ES =0000 0000000000000000 ffffffff 00c00000
> May 02 20:12:02 pve QEMU[1563981]: CS =0010 0000000000000000 ffffffff 00a09b00 DPL=0 CS64 [-RA]
> May 02 20:12:02 pve QEMU[1563981]: SS =0018 0000000000000000 ffffffff 00c09300 DPL=0 DS   [-WA]
> May 02 20:12:02 pve QEMU[1563981]: DS =0000 0000000000000000 ffffffff 00c00000
> May 02 20:12:02 pve QEMU[1563981]: FS =0000 0000000000000000 ffffffff 00c00000
> May 02 20:12:02 pve QEMU[1563981]: GS =0000 ffff8aab3bc00000 ffffffff 00c00000
> May 02 20:12:02 pve QEMU[1563981]: LDT=0000 0000000000000000 ffffffff 00c00000
> May 02 20:12:02 pve QEMU[1563981]: TR =0040 fffffe7730946000 00004087 00008b00 DPL=0 TSS64-busy
> May 02 20:12:02 pve QEMU[1563981]: GDT=     fffffe7730944000 0000007f
> May 02 20:12:02 pve QEMU[1563981]: IDT=     fffffe0000000000 00000fff
> May 02 20:12:02 pve QEMU[1563981]: CR0=80050033 CR2=00007f290c2ea8f5 CR3=000000014b1a4005 CR4=00370ef0
> May 02 20:12:02 pve QEMU[1563981]: DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000
> May 02 20:12:02 pve QEMU[1563981]: DR6=00000000fffe0ff0 DR7=0000000000000400
> May 02 20:12:02 pve QEMU[1563981]: EFER=0000000000000d01
> May 02 20:12:02 pve QEMU[1563981]: Code=90 66 0f 1f 00 48 89 f1 55 48 89 fd 53 48 c1 e9 20 48 8b 07 <89> 48 14 48 8b 07 89 70 10 48 8b 07 83 ce 08 89 70 10 31 db eb 12 bf c6 a7 00 00 83 c3 01

lspci -knn #PVE8.4.1
> 05:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 05)
>         Subsystem: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125]
>         Kernel driver in use: vfio-pci
>         Kernel modules: r8169

dmesg | grep -i 8125 #PVE8.4.1
> [    0.314390] pci 0000:05:00.0: [10ec:8125] type 00 class 0x020000 PCIe Endpoint
> [    1.047364] r8169 0000:05:00.0 eth0: RTL8125B, 00:e0:4c:68:06:c2, XID 641, IRQ 130

reset_method
> root@pve:~# cat /sys/bus/pci/devices/0000\:05\:00.0/reset_method #PVE8.4.1
> bus

dmesg | grep -i 8125 (RR 24.4.1 OK)
> root@rr:/opt/rr# dmesg | grep -i 8125
> [    0.363974] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [   12.557065] r8125: loading out-of-tree module taints kernel.
> [   12.558292] r8125 Ethernet controller driver 9.012.04-NAPI loaded
> [   12.579540] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [   12.581589] r8125  Copyright (C) 2023 Realtek NIC software team <nicfae@realtek.com> 
> [   18.817205] r8125: eth0: link up

dmesg | grep -i 8125 (junior 24.4.1)
> SynologyNAS> dmesg | grep -i 8125
> [    0.272016] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [    5.439981] r8125_v9 2.5Gigabit Ethernet driver 9.009.02-NAPI loaded
> [    5.472826] r8125_v9: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    5.476411] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com> 
> [   28.462229] r8125_v9: eth0: link up

dmesg | grep -i 8125 (RR 24.4.2 NG)
> root@rr:/opt/rr# dmesg | grep -i 8125
> [    0.353578] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [   12.542155] r8125: loading out-of-tree module taints kernel.
> [   12.544086] r8125 Ethernet controller driver 9.012.04-NAPI loaded
> [   12.571745] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [   12.573921] r8125  Copyright (C) 2023 Realtek NIC software team <nicfae@realtek.com> 
> [   18.775642] r8125: eth0: link up

dmesg | grep -i 8125 (junior 24.4.2)
> SynologyNAS> dmesg | grep -i 8125
> [    0.310002] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [    5.407419] r8125_v9 2.5Gigabit Ethernet driver 9.009.02-NAPI loaded
> [    5.433476] r8125_v9: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    5.437121] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com> 
> [   27.495494] r8125_v9: eth0: link up

不急着找解决方案，感谢dalao
看有没有用户遇见相同问题
随缘吧，毕竟修螃蟹卡的bug可能牵一发而动全身(早知就买i226了)

wjz304, 2025-06-02:
[image]

`kvm.ignore_msrs=1 kvm.report_ignored_msrs=0`  加到 RR_CMDLINE 里
试试

ctl26481, 2025-06-02:
> [image]
> 
> `kvm.ignore_msrs=1 kvm.report_ignored_msrs=0` 加到 RR_CMDLINE 里
> 
> 重启后生效
> 
> 试试

[image]

menu.sh編輯grub使用msrs cmdline(以及pve NUMA开启与否的组合)在RR Shell底下结果相同
(internal-error和error: kvm run failed Bad address)

ctl26481, 2025-06-02:
> 24.4.1 到24.4.2 RR 的系统文件并没什么更改

抱歉! 看来搞混版本了，再重新测试了遍
24.4.1 OK/24.4.2 OK，24.4.3 NG

dmesg | grep -i 8125 (RR 24.4.3 NG)
[    0.175273] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
[   12.451075] r8125: loading out-of-tree module taints kernel.
[   12.452696] r8125 Ethernet controller driver 9.012.04-NAPI loaded
[   12.474531] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[   12.476571] r8125  Copyright (C) 2023 Realtek NIC software team <nicfae@realtek.com> 
[   18.854340] r8125: eth0: link up 

dmesg | grep -i 8125 (junior 24.4.3，因应MD5的关係有手动上传/tmp/pat)
[    0.314889] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
[    5.450000] r8125_v9 2.5Gigabit Ethernet driver 9.009.02-NAPI loaded
[    5.475596] r8125_v9: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[    5.479216] r8125  Copyright (C) 2022 Realtek NIC software team <nicfae@realtek.com> 
[   28.294494] r8125_v9: eth0: link up

ctl26481, 2025-06-04:
感谢dalao! 测试结果如下

 * 相同的PVE internal-error和QEMU error: kvm run failed Bad address
 * 使用modprobe -r r8125移除驱动后可正常从RR shell下达poweroff指令
(init.sh显示Network devices not found! Please re execute init.sh after connecting to the network!)

dmesg | grep -i 8125 (RR shell 25.6.3 NG)

> [    0.440985] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [    0.812510] pci_bus 0000:01: resource 2 [mem 0x7000000000-0x700fffffff 64bit pref]
> [    3.996280] r8125: loading out-of-tree module taints kernel.
> [    3.998076] r8125 Ethernet controller driver 9.015.00-NAPI loaded
> [    4.023952] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    4.023966] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
> [   10.867822] r8125: eth0: link up

 * junior环境可正常从serial0下达poweroff指令
dmesg | grep -i 8125 (DS3622xs+ 7.2-72806 junior 25.6.3)

> [    0.266865] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [    5.738868] r8125 Ethernet controller driver 9.015.00-NAPI-RSS loaded
> [    5.764858] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    5.766415] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
> [    5.772781] proc_dir_entry 'net/r8125' already registered
> [    5.773519] Modules linked in: r8125(OE+) acpi_cpufreq(OE-) kvm_intel kvm irqbypass virtio_pci(OE) virtio(OE) virtio_ring(OE) r8125_txrx(OE) etxhci_hcd i8042(OE) processor button(OE) xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common mv14xx(O)
> [    5.778034]  [<ffffffffa0384017>] rtl8125_init_module+0x17/0x3a [r8125]
> [    5.801633] Error: Driver 'r8125' is already registered, aborting...
> [   19.472532] proc_dir_entry 'net/r8125' already registered
> [   19.474771] Modules linked in: r8125(OE+) r8168(OE) ixgbe(OE) be2net(OE) igb(OE) i2c_algo_bit(OE) e1000e(OE) dca vxlan ip6_udp_tunnel udp_tunnel vfat fat mdio(OE) cpufreq_conservative(OE) cpufreq_powersave(OE) cpufreq_performance(OE) cpufreq_ondemand(OE) cpufreq_stats(OE) crc_itu_t(OE) crc_ccitt sg(OE) psnap(OE) p8022(OE) llc(OE) ipv6 loop sha256_generic synorbd(O) synofsbd(O) lm90(OE) lm78(OE) lm75(OE) adm9240(OE) adm1031(OE) adm1021(OE) adt7475(OE) adt7470(OE) pcspkr(OE) pcspeaker(OE) uas usb_storage virtio_scsi(OE) atkbd(OE) virtio_console(OE) hvc_console(OE) pgdrv(O) evdev(OE) irqbypass virtio_pci(OE) virtio(OE) virtio_ring(OE) r8125_txrx(OE) etxhci_hcd i8042(OE) processor button(OE) xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common mv14xx(O) [last unloaded: synobios]
> [   19.499290]  [<ffffffffa0762017>] rtl8125_init_module+0x17/0x3a [r8125]
> [   19.519940] Error: Driver 'r8125' is already registered, aborting...
> [   19.533066] proc_dir_entry 'net/r8125' already registered
> [   19.533805] Modules linked in: r8125(OE+) r8168(OE) ixgbe(OE) be2net(OE) igb(OE) i2c_algo_bit(OE) e1000e(OE) dca vxlan ip6_udp_tunnel udp_tunnel vfat fat mdio(OE) cpufreq_conservative(OE) cpufreq_powersave(OE) cpufreq_performance(OE) cpufreq_ondemand(OE) cpufreq_stats(OE) crc_itu_t(OE) crc_ccitt sg(OE) psnap(OE) p8022(OE) llc(OE) ipv6 loop sha256_generic synorbd(O) synofsbd(O) lm90(OE) lm78(OE) lm75(OE) adm9240(OE) adm1031(OE) adm1021(OE) adt7475(OE) adt7470(OE) pcspkr(OE) pcspeaker(OE) uas usb_storage virtio_scsi(OE) atkbd(OE) virtio_console(OE) hvc_console(OE) pgdrv(O) evdev(OE) irqbypass virtio_pci(OE) virtio(OE) virtio_ring(OE) r8125_txrx(OE) etxhci_hcd i8042(OE) processor button(OE) xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common mv14xx(O) [last unloaded: synobios]
> [   19.557484]  [<ffffffffa079e017>] rtl8125_init_module+0x17/0x3a [r8125]
> [   19.577606] Error: Driver 'r8125' is already registered, aborting...
> [   23.187219] r8125: eth0: link up

 * DS环境可正常从serial0下达poweroff指令
dmesg | grep -i 8125 (DS3622xs+ 7.2.2-72806u3 25.6.3)

> [    0.318020] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [    5.865336] r8125 Ethernet controller driver 9.015.00-NAPI-RSS loaded
> [    5.891441] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    5.892994] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
> [    5.902271] proc_dir_entry 'net/r8125' already registered
> [    5.902996] Modules linked in: r8125(OE+) acpi_cpufreq(OE-) kvm_intel kvm irqbypass virtio_pci(OE) virtio(OE) virtio_ring(OE) etxhci_hcd r8125_txrx(OE) i8042(OE) processor button(OE) xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common mv14xx(O)
> [    5.907007]  [<ffffffffa038c017>] rtl8125_init_module+0x17/0x3a [r8125]
> [    5.924983] Error: Driver 'r8125' is already registered, aborting...
> [   62.508192] systemd[1]: Set hostname to <test8125>.
> [   69.616342] r8125: eth0: link up

ctl26481, 2025-06-04:
> 屏蔽掉 r8125_txrx ，使用原版 看看 DS 有问题吗

modprobe.blacklist=evbug,cdc_ether,**r8125_txrx** (从menu.sh > modprobe blacklist)
junior和DS环境那串`Modules linked in...`皆消失

 * junior环境可正常从serial0下达poweroff指令
dmesg | grep -i 8125 (DS3622xs+ 7.2-72806 junior 25.6.3)
> [    0.332957] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [    6.003508] r8125 Ethernet controller driver 9.015.00-NAPI loaded
> [    6.031907] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    6.034195] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
> [   22.048612] r8125: eth0: link up
> 

 * DS环境可正常从serial0下达poweroff指令
dmesg | grep -i 8125 (DS3622xs+ 7.2.2-72806u3 25.6.3)
> [    0.420130] pci 0000:01:00.0: [10ec:8125] type 00 class 0x020000
> [    6.070562] r8125 Ethernet controller driver 9.015.00-NAPI loaded
> [    6.111721] r8125: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
> [    6.117072] r8125  Copyright (C) 2024 Realtek NIC software team <nicfae@realtek.com> 
> [   62.167580] systemd[1]: Set hostname to <test8125>.
> [   70.695483] r8125: eth0: link up

ctl26481, 2025-09-19:
> 9?

on PVE 8.4.12/8.4.14:
RR 25.8.1 NG/NG
RR 25.9.4 NG/NG

wjz304, 2025-09-20:
你测试最新的pve9了吗

closed 2025-09-20 by ctl26481