RROrg/rr #8615: Issue with migration from DS925+(v1000nk) to other platforms

PeterSuh-Q3, 2025-05-04:
RR used the latest version 25.4.2.
When building the loader with a different model for migration from DS925+(v1000nk) to another platform and booting RR's DSM,
Migration cannot be done because the mount of the /dev/md0 partition failed.
The only migration is possible with DS1621+(v1000).
I am only guessing that it is possible because they are similar platforms.

[image]

The junior log of RR captured through the telnet port is as follows.

[rr_925p_to_6400_mig.log](https://github.com/user-attachments/files/20027353/rr_925p_to_6400_mig.log)
[rr_925p_to_923p_mig.log](https://github.com/user-attachments/files/20027351/rr_925p_to_923p_mig.log)
[rr_925p_to_918p_mig.log](https://github.com/user-attachments/files/20027352/rr_925p_to_918p_mig.log)
[rr_925p_to_1621p_mig.log](https://github.com/user-attachments/files/20027354/rr_925p_to_1621p_mig.log)

I thought that the cause of this problem was the absence of the /sbin/btrfs binary,
so I added the case of mshell separately and added btrfs-progs v4.0 as an Addon in advance, but there was no change in the issue.

https://github.com/PeterSuh-Q3/tcrp-addons/tree/main/abusyboxupgrade/auxfiles/usr

PeterSuh-Q3, 2025-05-06:
And, this is something that was confirmed a little while ago, but it seems that there is an issue with the btrfs.ko that is built into DS925+.
I have not modified any part related to btrfs in v1000nk in mshell.
If there is an issue with btrfs, there is a possibility that the storage pool will continue to be unstable.

[image]

PeterSuh-Q3, 2025-05-06:
In the case of Geminilake, btrfs.ko remains and does not disappear.

wjz304, 2025-05-07:
该问题似乎并不影响 型号 的迁移

PeterSuh-Q3, 2025-05-07:
由于这是 Synology 官方承认的错误，您可以在等待更新发布时使用此插件作为临时解决方案。 
MSHELL 已经应用，我计划只使用它直到下次更新修补为止。

apriliars3, 2025-05-12:
I am testing DS925+ on Proxmox with Intel 8500 and not works. Same issue with DS1525+. Thanks

[image]

[image]

wjz304, 2025-05-13:
> I am testing DS925+ on Proxmox with Intel 8500 and not works. Same issue with DS1525+. Thanks
> 
> [image]
> 
> [image]

日志并无异常

apriliars3, 2025-05-13:
> > I am testing DS925+ on Proxmox with Intel 8500 and not works. Same issue with DS1525+. Thanks
> > [image]
> > [image]
> 
> 日志并无异常

[logs.txt](https://github.com/user-attachments/files/20196391/logs.txt)