RROrg/rr #8938: Request to Include nct6775 and it87 Kernel Modules for Fan Sensor Support in Redpill DSM 7.2

TKASAKAWA, 2025-05-08:
Description
I am running Synology DSM 7.2-72806 on a custom hardware setup using Redpill Ver.25.5.1. The hardware is an NEC Express5800/T110i-S server with an Intel Xeon E3-1220v6 processor and 16GB RAM. I am attempting to monitor fan speed using Node Exporter and Prometheus, but the node_hwmon_fan_rpm metric is not available. This indicates that the fan sensors are not detected by the Linux hwmon subsystem.

Issue
The absence of node_hwmon_fan_rpm is due to the lack of appropriate fan controller drivers (nct6775 or it87) in the Redpill kernel. When attempting to load the nct6775 module, I receive the following error:

sudo modprobe nct6775
modprobe: ERROR: could not insert 'nct6775': No such device
This suggests that either:

The nct6775 kernel module is not included in the Redpill kernel.
The kernel cannot detect the Super I/O chip (likely Nuvoton NCT6775 or ITE IT87, common in NEC Express5800 servers).
Checking the hwmon subsystem confirms that only coretemp (CPU temperature) is detected, with no fan-related sensors:

ls /sys/class/hwmon
hwmon0

cat /sys/class/hwmon/hwmon0/name
coretemp

cat /sys/class/hwmon/hwmon*/fan*_input
<no output>
Node Exporter output also lacks node_hwmon_fan_rpm:

text

curl http://localhost:9100/metrics | grep 'node_hwmon_fan_rpm'
<no output>
Current Workarounds
To avoid modifying the kernel (due to the risk of breaking the system), I am considering alternatives like:

Using DSM’s native command (/usr/syno/sbin/synodisk --fan_speed) to scrape fan speed.
Using IPMI (ipmitool sensor) if supported by the hardware.
Using the DSM API (SYNO.Core.Hardware.Fan).
However, these workarounds require additional scripting and are less seamless than native hwmon integration via Node Exporter.

Request
To enable fan speed monitoring via the hwmon subsystem and Node Exporter, I kindly request the Redpill team to include the following kernel modules in the Redpill build for DSM 7.2:

nct6775: For Nuvoton Super I/O chips, commonly used in servers like NEC Express5800.
it87: For ITE Super I/O chips, as a fallback for other hardware.
Additional Information
DSM Version: 7.2-72806
Redpill Version: Ver.25.5.1
Hardware: NEC Express5800/T110i-S (Xeon E3-1220v6, 16GB RAM)

TKASAKAWA, 2025-05-08:
### Additional Issue: Missing `node_systemd_socket_current_connections` Metric
In addition to the fan sensor issue, I am unable to retrieve the `node_systemd_socket_current_connections` metric from Node Exporter. Running `curl http://localhost:9100/metrics | grep 'node_systemd_socket_current_connections'` returns no output, and Grafana shows "No data" for the query `node_systemd_socket_current_connections{instance="$node", job="$job"}`.

This suggests that the systemd collector (`--collector.systemd`) is either disabled or not fully supported in the Redpill kernel/systemd implementation. Checking systemd sockets (`systemctl list-sockets`) shows limited or no `.socket` units, possibly due to DSM 7's restricted systemd functionality.

**Request**:
- Enhance Redpill's systemd support to ensure `node_systemd_socket_current_connections` and related metrics are available.
- Verify that the systemd collector in Node Exporter can retrieve socket connection data in DSM 7 (Redpill Ver.25.5.1, kernel 4.4.180+).

**System Details**:
- systemd version: [output of `systemd --version`]
- Socket units: [output of `systemctl list-sockets`]

wjz304, 2025-05-08:
lsmod | grep -E 'it87|nct6775'

wjz304, 2025-05-13:
RR and DSM are two systems, DSM configuration items are not easy to change

TKASAKAWA, 2025-05-15:
Thank you for your hard work, wjz304! Knowing that you are the developer gives me confidence in the future. I will continue to follow this issue and look forward to the improvements you will bring. Keep up the great work!

# dmesg | grep -E 'it87|nct6775'
[   44.552618] it87: it87 driver version <not provided>

wjz304, 2025-05-15:
> Thank you for your hard work, wjz304! Knowing that you are the developer gives me confidence in the future. I will continue to follow this issue and look forward to the improvements you will bring. Keep up the great work!
> 
> # dmesg | grep -E 'it87|nct6775'
> [ 44.552618] it87: it87 driver version

当前 RR  中已经编译了 hwmon-vid it87 nct6683 nct6775 adt7470 adt7475 等驱动，如果硬件兼容 他们应该会被加载的，你检查一下他们为什么没有被加载