RROrg/rr #8944: BUG 最新版的AME无法降级导致无法激活

lingshengjiang898, 2025-05-08:
[image]
无法通过矿神方法激活

# DSM7.2 AME版本3.1.0-3005
curl -L http://code.imnks.com/ame3patch/ame72-3005.py | python

AME安装来源：https://github.com/007revad/Video_Station_for_DSM_722
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.5.2
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-05-08:
DSM7.2.2 中 AME 已无实际用途

lingshengjiang898, 2025-05-08:
> 报啥错
安装没有报错，激活失败报错

 curl http://code.imnks.com/ame3patch/ame72-3005.py | python
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1868  100  1868    0     0  22678      0 --:--:-- --:--:-- --:--:-- 22780
Patching
Checking whether patch is successful...
Patch is unsuccessful, retcode = 256

[image]

wjz304, 2025-05-08:
服务器问题

wjz304, 2025-05-08:
v25.5.3

closed 2025-05-08 by wjz304

lingshengjiang898, 2025-05-08:
> 好吧，知道了

辛苦大佬连发多版本