RROrg/rr #8947: BUG: QuickConnect ID register error and QC not working

azchoi, 2025-05-08:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.5.2
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  ds3622xs
 * version:  7.2.2-72806 Update 3

Issue:  

QuickConnect does not work after updating to 25.5.2.

When I entered the control panel, I checked that the QuickConnect ID in "External Access" is not registered, and when I try to reactivate it after disabling it, I receive network related errors and a continuous failure message.

So I rolled back to 25.4.2, and now it works normally again.

From this, it is presumed to be a bug in RR loader.

-------------------Chinese translation-----------------------------------------------------------
更新到25.5.2后，QuickConnect无法运行。

进入控制面板后发现"外部访问"中的QuickConnect ID无法注册，在禁用该功能后再次尝试激活时，会出现网络相关错误，并持续出现失败信息。

所以再次以25.4.2回滚，现在又恢复了正常动作。

由此推测是RR加载器的错误。

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

adiom123, 2025-05-08:
同样的问题

wjz304, 2025-05-08:
v25.5.3