RROrg/rr #9619: 脚本menu.sh中调用函数setStaticIP修改固定IP地址后，导致群晖无法启动。

lysgwl, 2025-05-16:
在脚本menu.sh中调用Settings menu中，设置静态IP，设置参数如下：
address: 192.168.2.14
netmask: 255.255.255.0
gateway: 192.168.2.1
dns: 192.168.2.1
导致群晖无法启动。

意图是，通过修改为固定IP地址，来设置群晖的网络地址。
刚开始通过设置可以进入群晖的安装界面（192.168.2.14:5000），离线安装完成DSM_SA6400_72806.pat，完成群晖的基本设置后，然后再次访问就不能进入了。什么原因导致呢？

wjz304, 2025-05-18:
test

lysgwl, 2025-05-19:
> test
测试了新版本rr-25.5.8.img，修改静态IP，群晖无法启动的问题依然存在。
静态地址：192.168.2.14
子网掩码：255.255.255.0
网关地址：192.168.2.1

[image]
尽管能进入以上的图片，但是无法在线下载。以下是启动的参数：
[image]

[image]

kirawang, 2025-05-20:
你试试使用同网段的电脑是不是可以访问，我也发现了这个问题。分析了一下，是安装DSM以后，进入DSM会丢失网关。只能在同网段去访问。我倾向是个BUG

wjz304, 2025-05-20:
本来就没支持首次安装，这很麻烦，不过找到了一个脱裤子放屁的办法，下个版本修复下吧

lysgwl, 2025-05-20:
> 本来就没支持首次安装，这很麻烦，不过找到了一个脱裤子放屁的办法，下个版本修复下吧

意思是，目前版本不支持，这个修改静态IP的功能是安装成功后，手动再次引导修改？那还不如进入群晖修改网络。

wjz304, 2025-05-20:
可以这么理解， 

本来就不打算写这个功能，因为从来都是建议 在上游 (路由侧) 设置静态IP，而不是在 下游 (终端侧) 设置。 
写这个功能只是为了引导能够进行编译。

wjz304, 2025-05-22:
https://github.com/RROrg/rr/releases/tag/25.5.9
test

lysgwl, 2025-05-23:
> https://github.com/RROrg/rr/releases/tag/25.5.9 test

测试了下，貌似可以了。太感谢了，犹如滔滔江水，绵绵不绝。

closed 2025-05-23 by wjz304