RROrg/rr #10141: BUG 风扇模式没有生效，不显示风扇调节菜单

lingshengjiang898, 2025-05-22:
请填写以下信息.  
Please fill in the following information.  

[image]

星云 b365 主板 i3-8300t，已经勾选插件，日志如下
# sensors-detect version 3.6.0
# System: OEM Synoden [To be filled by O.E.M]
# Board: OEM ITX-SC3
# Kernel: 5.10.55+ x86_64
# Processor: Intel(R) Core(TM) i3-8300T CPU @ 3.20GHz (6/158/11)

Running in automatic mode, default answers to all questions
are assumed.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8613E Super IO Sensors'                        Success!
    (address 0xa30, driver `to-be-written')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): 

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Using driver `i2c-i801' for device 0000:00:1f.4: Kaby Lake (PCH)

Next adapter: SMBus I801 adapter at f040 (i2c-0)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpc (i2c-1)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpb (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpd (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX B/DDI B/PHY B (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX D/DDI D/PHY D (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.

Driver `to-be-written':
  * ISA bus, address 0xa30
    Chip `ITE IT8613E Super IO Sensors' (confidence: 9)

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Note: there is no driver for ITE IT8613E Super IO Sensors yet.
Check https://hwmon.wiki.kernel.org/device_support_status for updates.

Do you want to generate /etc/sysconfig/lm_sensors? (YES/no): 
Copy prog/init/lm_sensors.service to /usr/lib/systemd/system
and run 'systemctl enable lm_sensors.service'
for initialization at boot time.

Install ENV: (You can find it in the boot interface.)  
 * DMI: 
 * CPU:   8100t
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

lingshengjiang898, 2025-05-22:
最新的 25.5.9 问题依旧

wjz304, 2025-05-22:
目测 it87 已驱动，没该主板，具体不知道

lingshengjiang898, 2025-05-22:
> 目测 it87 已驱动，没该主板，具体不知道

群晖 7.1 版本时候有这个菜单还能选择调整，后面升级到 7.2 之后就没了，不清楚啥原因

TKASAKAWA, 2025-05-23:
功能请求：关于在通用硬件上集成lm_sensors的IPMI BMC KCS驱动

大家好，

我目前在NEC Express5800/T110i-S (Xeon E3-1220v6)上使用Redpill Loader Ver. 25.5.9运行DSM 7.2-72806。

在设置硬件监控时，我注意到sensors-detect工具成功识别了ISA总线上的 IPMI BMC KCS芯片，但提示lm_sensors没有对应的驱动（“to-be-written”）。

sensors-detect的输出如下：

Driver `to-be-written':
  * ISA bus, address 0xca2
    Chip `IPMI BMC KCS' (confidence: 8)
尽管IPMI Web界面可以完美地进行全面管理，但如果未来能够将这种IPMI BMC KCS芯片的支持集成到Redpill环境中的lm_sensors里，那将是非常棒的。这样可以直接从操作系统更方便地监控硬件健康状况，如果必要的设备文件可用，或许也可以通过ipmitool等工具实现。

这不是一个紧急的请求，只是一个对未来发展方向的设想，如果可行的话。

感谢大家的辛勤工作！