RROrg/rr #10474: I'd like to receive a Telegram notification when I can connect to TTYD.

mytelegrambot, 2025-05-29:
I'd like to receive a Telegram notification when I can connect to TTYD.

On my N100 server, it takes over 20 minutes for TTYD to become accessible after I execute "RebootToLoader" from the Control Panel's Task Scheduler. Docker seems to have a significant impact on this delay.

So, it would be great to have a feature that sends a Telegram notification as soon as the TTYD connection is established.

wjz304, 2025-06-01:
[image]

[image]

mytelegrambot, 2025-06-01:
Thank you for incorporating my requests.