RROrg/rr #10685: BUG cpuinfo

wahabe, 2025-06-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
... 
Hello wjz304

There is an error in the processor reading.
Get reading Threads not Cores

I am testing DS925+ and  RS2825RP+ on i7-14700K

If you need any reports, do not hesitate

https://xpenology.com/forum/topic/71132-develop-and-refine-the-ds925v1000nk-loader/page/2/#findComment-479067

[image]

wjz304, 2025-06-02:
[image]
?

closed 2025-06-03 by wjz304

wahabe, 2025-06-03:
efficient-cores work on HD6500 , SA6400 ?

[image]

[image]

wjz304, 2025-06-03:
`cat /proc/cpuinfo`

wahabe, 2025-06-03:
No one tested efficient-cores ?