RROrg/rr #10690: BUG

taotaolife, 2025-06-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  i5-1235U
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.6.1
 * addons:  
 * modules:  all
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2.2-72806 Update 3

风扇驱动hwmon-vid.ko、it87.ko会导致资源监控卡死，存储管理器找不到硬盘，取消这两个驱动后资源监控正常，存储管理器也能找到盘了，但无法通过控制面板调速了

[image]

[image]

[image]

wjz304, 2025-06-02:
[image]
test

taotaolife, 2025-06-02:
> [image] test

添加参数后问题依旧

wjz304, 2025-06-04:
那就不兼容呗，

taotaolife, 2025-06-16:
> 那就不兼容呗，

最新版本25.6.5已经没有该问题了

taotaolife, 2025-06-16:
[image]

[image]

[image]

风扇也能正常调速了

> 那就不兼容呗，

closed 2025-07-23 by wjz304