RROrg/rr #10691: 系统不能识别HBA卡和连接的硬盘

981041409, 2025-06-02:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  8100T
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.5.10/25.5.9
 * addons:  LSI9300-8i/浪潮3008
 * modules:  
 * lkms:  

DSM: 
 * model:  DS3622xs/DS1823xs/918
 * version:  7.2.2/7.2.1

Issue:  
配置：主板：华擎B365Mpro4
           处理器：8100T
           内存：16G
           硬盘：希捷16T*1
                      HC550*4
           通过引导直接（不是虚拟机）安装群辉后不能够识别直通卡和连接的所有硬盘
logs:  
[

[image]

](url)
(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-06-03:
请咨询主板商和 HBA卡商

AuxXxilium, 2025-06-03:
> 请咨询主板商和 HBA卡商

for your information: it was working for all release before 25.5.10 - in this release where the module for mpt changed, this cause this issue.

wjz304, 2025-06-04:
lspci 显示所有 pci 设备, 有没有驱动都不影响 lspci 的显示，请咨询主板商和 HBA卡商,  与引导无关，该 issue 关闭。

closed 2025-06-04 by wjz304