RROrg/rr #11057: BUG

laobamac, 2025-06-05:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Lenovo
 * CPU:  E3-1231V3
 * NIC:  (pid & vid) 812510EC

RR version: (You can find it in the update menu.)  
 * RR:  25.5.4
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  918+
 * version:  7.2u3

Issue:  Can't run Linux in VMM directly.Linux virtual machines can only use CPU compatibility mode or disable the AVX instruction set in GRUB, which is not an issue for Windows. It seems that the CPU model of the host is different from that of the 918, which causes some features to fail to be transparently transmitted. 

logs:  VMM didn't have any logs

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-06-06:
This requires checking whether the Linux you installed in the vmm supports AVX

laobamac, 2025-06-06:
Yep,both the Processor and System support AVX (CentOS7/8/9/10，Ubuntu 18/22/24).Reason of this problem is CPU Model Mistakes, see https://www.openos.org/threads/6-17-ds3617-vmm-linux-kernel-panic.797/ .Could RR fix this issue on DSM918+？Thx.

PS:J2900 processor on DSM918 with RR worked VMM perfectly,because Jxxx CPU.And I use 3622XS on E3-1231v3, VMM become well-worked.

wjz304, 2025-06-06:
[image]

[image]

[image]

test

wjz304, 2025-06-06:
It is recommended to disable it in the grub of the VMM virtual machine.

closed 2025-06-08 by wjz304