RROrg/rr #11090: 症状描述：PC扬声器（蜂鸣器）不工作。

TKASAKAWA, 2025-06-06:
症状描述：PC扬声器（蜂鸣器）不工作。

当前系统已更新至 Redpill Loader (RR) 25.6.3 版本，PC扬声器功能停止。
尽管 pcspkr 内核模块已通过 lsmod 命令确认加载，且 /sys/bus/platform/drivers/pcspkr/ 驱动目录也存在并处于激活状态，但相应的平台设备路径 /sys/devices/platform/pcspkr/ 却不存在。
这导致了 /dev/input/eventX 等蜂鸣器设备文件无法被创建。
由于缺少设备文件，beep 命令也无法正常工作。

在此之前，PC扬声器功能是正常的。问题是在今天更新到 RR 25.6.3 版本后才出现的。
这表明可能不是一个传统的软件错误，而是在新版本或特定编译配置下，pcspkr 驱动无法正确识别或注册硬件设备，或者可能需要某个未被识别的强制性模块。

英語 (English) - Final Revised Version:

Symptom Description: PC Speaker (Beeper) is not functioning.

The system is currently updated to Redpill Loader (RR) version 25.6.3, and the PC speaker functionality has ceased.
Although the pcspkr kernel module is confirmed to be loaded via the lsmod command, and the driver directory /sys/bus/platform/drivers/pcspkr/ exists and is active, the corresponding platform device path /sys/devices/platform/pcspkr/ is absent.
This absence prevents the creation of beeper device files such as /dev/input/eventX.
Consequently, the beep command is unable to produce any sound due to the lack of these device files.

Previously, the PC speaker functionality was working correctly. The issue specifically appeared today after updating to RR version 25.6.3.
This suggests it might not be a conventional software bug, but rather a scenario where the pcspkr driver fails to correctly recognize or register the hardware device under the new version or specific build configuration, or possibly there's an unrecognized mandatory module dependency.

wjz304, 2025-06-06:
lsmod | grep pcsp

wjz304, 2025-06-06:
> ```
> # lsmod | grep pcsp
> pcspkr                 16384  0 
> ```

grep pcsp /usr/rr/backup/p1/user-config.yml

wjz304, 2025-06-06:
```
root@test:~# lsmod | grep pcsp
pcspkr                 16384  0 
pcspeaker              16384  0 

root@test:~# grep pcsp /usr/rr/backup/p1/user-config.yml
  pcspeaker: ""
  pcspkr: ""
```
Check if pcspeaker is disabled by you, it should load automatically.