RROrg/rr #11179: DSM7.2环境下，USB移动设备存储如何设置为群晖的内置硬盘呢？

lysgwl, 2025-06-07:
在PVE宿主机环境下，安装DSM虚拟机，rr引导镜像中，设置以下参数：
esataportcfg="0xff000"
usbportcfg="0x300000"
internalportcfg="0xfff"

并且查看 /etc.defaults/synoinfo.conf 文件，以上配置选项确认修改，但是在DSM环境下，USB存储设备并未识别为内置硬盘，什么原因导致呢？请大佬帮忙解答，谢谢。

ctl26481, 2025-06-08:
试试计算器
先从console/shell看USB硬盘辨识的盘符，再将盘符改为你需要的模式
https://github.com/RROrg/rr/issues/226

lysgwl, 2025-06-08:
> 试试计算器 先从console/shell看USB硬盘辨识的盘符，再将盘符改为你需要的模式 [#226](https://github.com/RROrg/rr/issues/226)

好的，大佬，是不是7.2的版本在/etc.defaults/synoinfo.conf 中修改端口映射的方法已经失效了吧？

lysgwl, 2025-06-08:
> [image]

根据提示操作，生成了model.dts文件，意思是在model.dts配置文件中，把usb设备信息修改为internal_slot标识设备，然后重新编译，是这意思吗？理解是不到位？

信息如下：
[model.txt](https://github.com/user-attachments/files/20642679/model.txt)

wjz304, 2025-06-08:
[image]

lysgwl, 2025-06-08:
> [image]

[image]

[image]

[image]

[image]

wjz304, 2025-06-08:
[image]

[image]

closed 2025-06-08 by wjz304