RROrg/rr #11197: DSM 7.2-72806 的 SMART 和历史记录选项变灰的问题

TKASAKAWA, 2025-06-09:
尊敬的 Synology 支持团队，

您好！我是使用 DSM 7.2-72806 DiskStation DS1525+ 的用户，我的设备使用的是 NEC Express5800/T110i-S 机箱，而不是原装机箱。我注意到健康状态页面中的 SMART 和历史记录选项是灰色的，无法使用。我的硬盘是 WD WD20PURZ 和 Samsung SSD ，还使用了 Redpill 环境（Ver.25.6.4）。请问这可能是由于非原装机箱或环境设置导致的吗？希望您能提供一些温和的建议，帮助我解决这个问题。谢谢！

Hello! I’m a user of the DSM 7.2-72806 DiskStation DS1525+, and my device uses an NEC Express5800/T110i-S chassis instead of the original one. I’ve noticed that the SMART and history options on the health status page are grayed out and unavailable. My drives are WD WD20 and Samsung SSD , and I’m using the Redpill environment (Ver.25.6.4). Could this be due to the non-standard chassis or environment settings? I’d really appreciate some gentle advice to help me resolve this issue. Thank you!

wjz304, 2025-06-11:
addons 中 你是否开启了 smartctl?

TKASAKAWA, 2025-06-12:
wjz304 您好，

首先，我想强调的是，我理解我的情况可能是一个相当罕见的案例。鉴于此功能对许多其他使用插件/附加组件的用户来说非常重要，我强烈希望您能继续维持并支持这种插件/附加组件的运行模式。

现在，关于我这边的问题，我有一个非常重要的发现！我尝试了 smartctl -a -d sat /dev/sata1 命令，这次成功获取了完整的 SMART 信息：

smartctl 7.5 2025-04-30 r5714 [x86_64-linux-5.10.55+] (localbuild)
...
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
SMART overall-health self-assessment test result: PASSED
... (以下是完整的 SMART 属性和日志)
这意味着硬盘本身是支持 SMART 并且功能正常的。问题在于 smartctl 命令需要明确指定 -d sat 选项才能正确与硬盘通信并获取 SMART 信息。

这块硬盘是 WD 的 Purple CCTV 系列硬盘。这类硬盘的主要流通渠道是面向系统集成商、安防设备供应商和家电零售商的法人部门等，主要作为 NVR/DVR 系统的一部分或替换件进行批量供应，因此在零售市场不常见。 此前无论使用 DSM 标准的 smartctl 6.5 还是 Entware 的 smartctl 7.5，在不加 -d sat 选项的情况下都显示 SMART support is: Unavailable。但加上 -d sat 后，完整信息就被成功读取了。这表明问题并非出在 smartctl 命令的版本本身。

我的情况可能是个例：我在另一台 HP 主机上安装同样的 DSM 后，SMART 信息可以正常显示且没有灰色问题。这似乎进一步证实，我的 NEC Express5800/T110i-S 服务器（使用 Intel 芯片组）上的 SATA 控制器，与当前 RR 25.6.4（内核版本 5.10.55+）在获取 SMART 信息时存在特定的兼容性问题，导致需要 -d sat 选项。

值得一提的是，据我所知，WD 的硬盘在 Synology 的官方兼容性列表中通常很少出现，并且往往被标记为非兼容。然而，这并非表示硬盘本身不兼容或有技术问题，更像是 Synology 的一种品牌策略或支持范围考量。 既然 smartctl -a -d sat 命令能够成功读取 SMART 信息，这进一步证实了硬盘在技术层面上是完全兼容 SMART 的。

由于在 DSM 的 GUI 中，所有硬盘的 SMART 信息依然是灰色且不可用，这强烈表明 DSM 内部用于获取 SMART 信息的服务，在与我的 NEC Express5800/T110i-S 服务器上的 Intel SATA 控制器交互时，可能没有自动或正确地使用 -d sat 这样的设备类型选项。

您认为是否有可能修改 DSM 内部调用 smartctl 的脚本或配置，使其包含 -d sat 选项，以解决 SMART 信息的显示问题？

再次感谢您的持续帮助，这个发现应该能大幅度缩小问题的范围！

First and foremost, I understand that my situation might be quite a rare case. Given that this functionality is crucial for many other users relying on plugins/addons, I strongly hope you will continue to maintain and support this addon-based operational model.

Now, regarding my specific issue, I have a very important update! I tried the command smartctl -a -d sat /dev/sata1, and this time, I successfully retrieved the complete SMART information:

smartctl 7.5 2025-04-30 r5714 [x86_64-linux-5.10.55+] (localbuild)
...
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
SMART overall-health self-assessment test result: PASSED
... (full SMART attributes and logs follow)
This clearly indicates that the hard drive itself supports SMART and is functioning correctly. The issue was that the smartctl command required the explicit -d sat option to communicate properly with the drive and retrieve SMART information.

This drive is a WD Purple CCTV series drive. Its primary distribution channels are typically corporate clients such as system integrators, security equipment suppliers, and corporate divisions of electronics retailers, primarily supplied in bulk for NVR/DVR systems or as replacement parts, making them uncommon in the retail market. Previously, both the DSM standard smartctl 6.5 and the Entware smartctl 7.5 showed SMART support is: Unavailable without the -d sat option. However, with -d sat added, the full information was successfully read. This indicates that the problem is not with the smartctl command version itself.

My case seems to be a rare one: I have another HP chassis running the same DSM where SMART information displays normally without being grayed out. This further confirms that there's a specific compatibility issue between the SATA controller on my NEC Express5800/T110i-S server (with its Intel chipset) and the current RR 25.6.4 (kernel version 5.10.55+), requiring the -d sat option for SMART retrieval.

It's also worth noting that, to my knowledge, Western Digital drives are generally rarely listed on Synology's official compatibility list and are often marked as unsupported. However, this does not imply a technical incompatibility of the drive itself, but rather seems to be related to Synology's brand strategy or scope of support. Given that the smartctl -a -d sat command was able to successfully read SMART information, this further confirms that the drive is fully SMART-compatible at a technical level.

Since SMART information for all drives remains grayed out and unavailable in the DSM GUI, this strongly suggests that the internal DSM service responsible for fetching SMART information might not be automatically or correctly using a device type option like -d sat when interacting with the Intel SATA controller on my NEC Express5800/T110i-S server.

Do you think it's possible to modify the scripts or configurations within DSM that call smartctl to include the -d sat option, in order to resolve the SMART information display issue?

Thank you again for your continued assistance; this discovery should significantly narrow down the problem!

wjz304, 2025-06-12:
smartctl 插件就是做这个用的

wjz304, 2025-08-27:
try 25.8.1

TKASAKAWA, 2025-08-27:
Thank you very much for fixing . I truly appreciate your prompt support and dedication. Your help made a real difference.
非常感谢您修复了这个问题。我由衷地感激您迅速的支持和付出，您的帮助真的非常重要。