RROrg/rr #11420: BUG：SA6400型号official内核下DG1无法驱动

JiangGeek, 2025-06-12:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  i7-8700k
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.6.5
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  sa6400
 * version:  DSM 7.2.2-72806 Update 3

Issue:  
official内核下DG1无法驱动，但测试custom内核下DG1能够正常驱动，看目录：/lib/firmware/i915下固件也是有的
logs:  
dmesg | grep i915
[    7.005875] i915 0000:04:00.0: Force probing unsupported Device ID 4908, tainting kernel
[    7.007418] i915 0000:04:00.0: [drm] VT-d active for gfx access
[    7.007955] i915 0000:04:00.0: [drm] Local memory IO size: 0x00000000fb800000
[    7.008540] i915 0000:04:00.0: [drm] Local memory available: 0x00000000fb800000
[    7.031341] i915 0000:04:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=none
[    7.033100] i915 0000:04:00.0: [drm] Finished loading DMC firmware i915/dg1_dmc_ver2_02.bin (v2.2)
[    7.033522] i915 0000:04:00.0: [drm] *ERROR* GT0: GuC firmware i915/dg1_guc_70.bin: fetch failed -ENODEV
[    7.034612] i915 0000:04:00.0: [drm] GT0: GuC firmware(s) can be downloaded from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[    7.039885] RIP: 0010:intel_context_init+0x58/0x1a0 [i915]
[    7.048349]  [<ffffffffa027598e>] ? intel_context_create+0x2e/0x90 [i915]
[    7.048942]  [<ffffffffa027796e>] ? intel_engine_create_pinned_context+0x1e/0x150 [i915]
[    7.049637]  [<ffffffffa02780aa>] ? intel_engines_init+0x32a/0x9d0 [i915]
[    7.050229]  [<ffffffffa027fef0>] ? execlists_unwind_incomplete_requests+0x20/0x20 [i915]
[    7.050932]  [<ffffffffa0285946>] ? intel_gt_init+0x146/0x330 [i915]
[    7.051496]  [<ffffffffa02c8f74>] ? i915_gem_init+0x1d4/0x220 [i915]
[    7.052050]  [<ffffffffa0253541>] ? i915_driver_probe+0x721/0xc10 [i915]
[    7.057996]  [<ffffffffa04d9027>] ? __init_backport+0x27/0x80 [i915]
[    7.062464] Modules linked in: kvm_intel(OE) kvm irqbypass coretemp(OE) rtc_cmos(OE) i2c_i801(OE) e1000e(OE+) i915(OE+) check_signature(OE) i2c_smbus(OE) drm_buddy(OE) intel_gtt(OE) drm_display_helper(OE) drm_kms_helper(OE) ttm(OE) drm(OE) drm_mipi_dsi(OE) drm_panel_orientation_quirks(OE) i915_compat(OE) dmabuf(OE) i2c_algo_bit(OE) thermal(OE+) evdev(OE) button(OE) acpi_cpufreq(OE) xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common
[    7.149767] RIP: 0010:intel_context_init+0x58/0x1a0 [i915]

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

wjz304, 2025-06-12:
ls -al /usr/lib/firmware/i915/dg1_guc_70.bin

JiangGeek, 2025-06-12:
> ls -al /usr/lib/firmware/i915/dg1_guc_70.bin md5sum /usr/lib/firmware/i915/dg1_guc_70.bin

[image]

JiangGeek, 2025-07-01:
@wjz304 大佬有办法修复吗，测试皮蛋熊最开始的驱动是能使用的： https://blog.kkk.rs/archives/28#%E5%85%ADrr%E9%85%8D%E7%BD%AE

JimiZhou, 2025-07-17:
遇到了同样的问题，安装上DG1之后，official内核引导后黑屏无法搜到IP，custom可以引导进入但是VMM虚拟机又无法使用了

wjz304, 2025-07-23:
https://github.com/RROrg/rr/releases/tag/25.7.0

JiangGeek, 2025-07-24:
> https://github.com/RROrg/rr/releases/tag/25.7.0
测试新版本还是无法直接驱动DG1，内核日志还是一样的“ERROR GT0: GuC firmware i915/dg1_guc_70.bin: fetch failed -ENODEV”