RROrg/rr #12764: After updating to 25.7.0

eragon4k, 2025-07-24:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.7.0
 * addons:  default
 * modules:  default
 * lkms:  

DSM: 
 * model:  DS3622xs
 * version:  7.2.2-72806 update 4

Issue:  After updating to RR25.7.0 and DSM 7.2.2-72806 update 4. RTL8125B NIC is not detected anymore.

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

eragon4k, 2025-07-25:
Roll back to 25.6.5 fixed the problem for now.

eragon4k, 2025-07-30:
This happens time to time when the nic driver gets updated, so I have one intel and one realtek as a precaution.

Fyred, 2025-08-03:
早点来翻issues就好了，hyper-v虚拟的DSM，升级25.7.0后启动DSM宿主机死机，然后网卡消失，我还以为硬件问题，甚至一度怀疑天气太热网卡挂了。。。折腾了一天

wjz304, 2025-08-04:
https://github.com/RROrg/rr/releases/tag/25.8.0