RROrg/rr #12945: BUG,SA6400在custom下，Virtual Machine Manager虚拟机有问题

zhoyed, 2025-07-28:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  Dell Inc. PowerEdge R720xd/0T0WRN, BIOS 2.9.0 12/06/2019
 * CPU:  Intel® Xeon(R)™ E5-2630
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  v25.6.5
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400(CUSTOM)
 * version:  7.2.2-72806

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
... 
SA6400在custom下，Virtual Machine Manager虚拟机有问题，Virtual Machine Manager映像里面新增DSM映像总是卡到40%不动，从ABB导入的虚拟机一直显示正在启动，启动不了。有时候新建的虚拟机，在重启RR后，虚拟机也无法启动了，换了好几台电脑都是一样的问题。在official模式下，虚拟机功能一切正常。

wjz304, 2025-08-21:
目前解决不了

Eyre5300, 2025-10-06:
遇到了同样的问题，启动不起来

wjz304, 2025-11-01:
【腾讯文档】关于 custom SA6400 需求
https://docs.qq.com/form/page/DYUZrcFdGaHJ0TGVv

closed 2025-12-17 by wjz304