RROrg/rr #13778: Error message appears at startup

wahabe, 2025-08-06:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  

Hi wjz304, I'm glad you're back.

Error message appears at startup

Message appears from the beginning of the release rr-25.6.4 to rr-25.7.0

Note: The error only appears when using a card HBA

[image]

...

wjz304, 2025-08-14:
7.0 和8.0 并没做硬盘相关的修改
我看看的把

wjz304, 2025-09-06:
Note: The error only appears when using a card HBA

bootloader disk on HBA ?

wahabe, 2025-09-07:
no 

bootloader disk on Motherboard
The message appears even when using a flash memory.

Now, after updating to rr 8.1, another message appears even without using a HBA

i tried with 3 pc and the massage appeared on all 3 of them

tomorrow i will Send picture of the Message

wjz304, 2025-09-07:
https://github.com/RROrg/rr/releases/tag/25.9.2