RROrg/rr #15156: rr-25.9.1引导 ipv6网络BUG

orgx2812, 2025-09-05:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  H110
 * CPU:  g4560
 * NIC:  (pid & vid) 8168

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  7.2.2-72806 update 4

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
... 
网络界面管理 启用 Open vSwitch选项，或者安装Virtual Machine Manager套件使用网络就会出现ipv6无法正常使用！

[image]

[image]

网络界面管理 禁用 Open vSwitch选项 ipv6网络恢复正常

[image]

[image]
初次反馈，不知道是否符合格式。

wjz304, 2025-09-06:
```
curl  checkip.synology.com
curl  checkipv6.synology.com
cat /etc/resolv.conf
```

orgx2812, 2025-09-07:
> <img alt="图像" width="777" height="468" src="https://private-user-images.githubusercontent.com/5615843/486500121-c4f60554-894b-46fb-beb9-8464b7595f9c.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTcyNDY5MjYsIm5iZiI6MTc1NzI0NjYyNiwicGF0aCI6Ii81NjE1ODQzLzQ4NjUwMDEyMS1jNGY2MDU1NC04OTRiLTQ2ZmItYmViOS04NDY0Yjc1OTVmOWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MDdUMTIwMzQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGU5YTc2ODA4OTAxMGE4ZWYxM2I4MmRmZjRjZDRhNWUwZDUxNTE3NjU5Nzc0YWMyMzc1MWRjN2EyNWZkYTE0MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.yXz7s1pFy8Qe46VYc3xJgJV8lRUZeS9CeOjgRj4BPj4">

感谢 佬 这个确实可以解决ipv6问题，白裙不需要这样设置。有解决办法就好，辛苦 佬了。

wjz304, 2025-09-09:
https://github.com/RROrg/rr/releases/tag/25.9.3

orgx2812, 2025-09-10:
> https://github.com/RROrg/rr/releases/tag/25.9.3

第一时间更新了引导，网络设置好像没什么变化，是不是要重新安装DSM 才可以？

> 因为 eth0 一个， ovs_eth0 一个

这个疑问，按这个说那ipv4是否也应该有两个网关，但是ipv4只有一个网关而且网络是正常的。

orgx2812, 2025-09-13:
大佬 我尝试使用arpl-i18n引导，全部同样套件环境ipv6网络恢复正常。
https://github.com/syno-community/arpl-i18n/releases/download/23.10.4/arpl-i18n-23.10.4.img.zip

[image]

[image]

[image]

我感觉可以按这个尝试应该可以彻底解决网络问题。

wjz304, 2025-09-19:
test https://github.com/RROrg/rr/releases/tag/25.9.4

orgx2812, 2025-09-20:
> test https://github.com/RROrg/rr/releases/tag/25.9.4

已测试在rr-25.8.1 上升级至rr-25.9.4引导，物理机还是复现上述问题，物理机回退至rr-25.8.1引导网络恢复正常。rr-25.9.4使用虚拟机安装没有ipv6网络问题。

wjz304, 2025-09-20:
[eudev.zip](https://github.com/user-attachments/files/22439961/eudev.zip)

test

Unzip and get eudev.addon

[image]

[image]

rebuiild and boot

orgx2812, 2025-09-21:
> [eudev.zip](https://github.com/user-attachments/files/22439961/eudev.zip)
> 
> test
> 
> Unzip and get eudev.addon
> 
> <img alt="Image" width="384" height="292" src="https://private-user-images.githubusercontent.com/5615843/486987061-6b28650d-7de0-49d9-aa98-a7c171186994.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg0MTY0OTMsIm5iZiI6MTc1ODQxNjE5MywicGF0aCI6Ii81NjE1ODQzLzQ4Njk4NzA2MS02YjI4NjUwZC03ZGUwLTQ5ZDktYWE5OC1hN2MxNzExODY5OTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MjFUMDA1NjMzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGUxMzljMTRiZmNiMWQ1M2Y2NjA4MzhjOGI5OGQ4ZGQzNzkxOTRlZjU5MjdlMjhhMTdjZTg2NmI3ZGYyMzNhZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.JVf8uaUfXF8YzqJpKQl7Aan9kLF_6ci1sGd36PmLaSU"> <img alt="Image" width="496" height="144" src="https://private-user-images.githubusercontent.com/5615843/486987265-8a5a8d7b-1725-4530-b704-75d55db0c08e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg0MTY0OTMsIm5iZiI6MTc1ODQxNjE5MywicGF0aCI6Ii81NjE1ODQzLzQ4Njk4NzI2NS04YTVhOGQ3Yi0xNzI1LTQ1MzAtYjcwNC03NWQ1NWRiMGMwOGUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MjFUMDA1NjMzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGRmODgyNjM4YjgwOGQxMGU1ZTM3MzkwOGZmZTYyOWJhOGYxMGJjNzM5MGMxNTZkYjMxMzc3M2I4YjNjODE0ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.by1DQWC4M4wmspTjQnpiXuMkW94E7gyrfIh69nBI6O4">
> rebuiild and boot

Thanks to ipv6 network problem is solved。

wjz304, 2025-09-24:
https://github.com/RROrg/rr/releases/tag/25.9.5

closed 2025-09-24 by wjz304