RROrg/rr #15479: BUG - docker dns issue

Kali-, 2025-09-08:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  25.9.2
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  
after upgrading to 25.9.2 i had serious issue with docker and all container was complaining about issue with the dns as they was unable to talk with 127.0.0.11 port 53
i have tried to move some of the container to host network and the issue sames to be mitigated but not completly resolved, so i guess there is something else broken in the net stack (not only the dns)
after update i noticed also some odd issue in the network settings, all interface LAN 1 to 4 which are part of 2 Bond are visibile in the network settings (on 25.8.1 i only see Bond1-2) and my wifi can't self connect anymore

rolled back to 25.8.1 as solved all the issue
 
logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

VincentBFG, 2025-09-08:
I post it in other Issue.

But it looks like the problem is very close.
Problems with network interfaces.
I haven't done a rr rollback yet.

After updating the bootloader to version 25.9.2, the running system stopped receiving addresses via DHCP on two network cards of the motherboard. If I connect the cable to an additional network card, the address is received. Before this update, the problem was with the Bond SLB interface, the network ports assigned to it dropped out. I had to delete the Bond interface and create it again. Now it is impossible to create a Bond, the interface is configured, but the interface is not created and the ports that were added to it disappear.

I also did Force reinstall DSM, it didn’t help.

Addition:
The web interface does not allow access to Regional settings. It gives the error - Connection failed, check your network settings.

[image]

crazyb123, 2025-09-09:
@wjz304 采用上文的[eudev.zip](https://github.com/user-attachments/files/22222240/eudev.zip)之后，创建bond，重启后一切正常，和以前一样，网口1和网口2隐藏，只剩bond。感谢！

wjz304, 2025-09-09:
https://github.com/RROrg/rr/releases/tag/25.9.3

closed 2025-09-09 by wjz304

Kali-, 2025-09-10:
i think i found the issue:

```
# netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG        0 0          0 ovs_bond0
10.255.0.0      0.0.0.0         255.255.255.0   U         0 0          0 ztbpaicxdw
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 eth2
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 eth1
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 eth3
169.254.0.0     0.0.0.0         255.255.0.0     U         0 0          0 ovs_bond1
**192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0**
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 ovs_bond0
192.168.200.0   0.0.0.0         255.255.255.0   U         0 0          0 ovs_eth80
```
there's a wrong route on eth0, which is part of bond0
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0

i have removed it and network looks to be resolved

`route del -net 192.168.1.0 gw 0.0.0.0 netmask 255.255.255.0 dev eth0`