RROrg/rr #17173: 无网络

jackson-lx, 2025-09-25:
我使用的是hp gen10，编译，问题有两个，不管怎么弄，套件中心上不了网，通过ssh上去，是可以上网的。
第二个问题就是，讲gen10 的网卡设置为端口汇聚，（原来7.1的版本没有出现在过），配置好所有参数，只要开机重新，就没有网络了，找不到，把硬盘拆下来到windows上看，就是显示分区损坏。

jackson-lx, 2025-09-26:
> <https://user-images.githubusercontent.com/5615843/235939097-6798da58-24fd-44cc-9970-c8d2f9609704.jpg>
>
> 请填写以下信息.
> Please fill in the following information.
>
> Install ENV: (You can find it in the boot interface.)
>
>    - DMI:
>    - CPU:
>    - NIC: (pid & vid)
>
> RR version: (You can find it in the update menu.)
>
>    - RR:
>    - addons:
>    - modules:
>    - lkms:
>
> DSM:
>
>    - model:
>    - version:
>
> Issue:
>
> logs:
>
> (## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)
> (## Because the log contains some sensitive information such as SN/MAC,
> please delete them when providing the complete file. Of course, you can
> also send it to my email. ##)
> ...
>
> (请先看一下#173 <https://github.com/RROrg/rr/issues/173>、#175
> <https://github.com/RROrg/rr/issues/175>、#226
> <https://github.com/RROrg/rr/issues/226> 的内容)
> (Plz review the content of #173 <https://github.com/RROrg/rr/issues/173>,
> #175 <https://github.com/RROrg/rr/issues/175>, #226
> <https://github.com/RROrg/rr/issues/226> first)
> ...
>
> (如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)
> (If you just say XXX doesn't work without providing any details, I can
> only say thank you for your feedback.)
> ...

所有参数这个都有了，问题是端口汇聚做好了，第一次是可以访问的，然后就是重启了就访问不了，交换机上显示协商失败

jackson-lx, 2025-09-26:
作者你好，经过重新安装测式，端口汇聚在安装第一次时，成功生效，只要重新启动，端口汇聚就失败了，找不到机器，这个时候把端口汇聚（交换机上的删除掉），用单网口进行连接，启动（true与flase）都测试了，开机看得到ip，电脑能ping通，群晖助手找不到群晖，过一会，在什么都没有动的情况下，ping通的ip不通了，后面就更找不到群晖了。因为是用的gen10，没有调试接口。

wjz304, 2025-09-26:
> 所有参数这个都有了，问题是端口汇聚做好了，第一次是可以访问的，然后就是重启了就访问不了，交换机上显示协商失败
> 
> github-actions[bot] ***@***.***> 于2025年9月25日周四 15:22写道：
> […](#)

https://github.com/RROrg/rr/releases/tag/25.9.6

closed 2025-09-26 by wjz304