RROrg/rr #18166: BUG，安装完dsm后一启动就自动关机。

646185642, 2025-09-27:
请填写以下信息.  
Please fill in the following information.  

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  (pid & vid)

RR version: (You can find it in the update menu.)  
 * RR:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(## 因为 log中存在 SN/MAC 等一些敏感信息, 当提供完整文件时请自行抹除他们, 当然你也可以发送到我的邮箱. ##)  
(## Because the log contains some sensitive information such as SN/MAC, please delete them when providing the complete file. Of course, you can also send it to my email. ##)    
...

(请先看一下#173、#175、#226 的内容)  
(Plz review the content of #173, #175, #226 first)  
...

(如果你只是说 XXX 不能用, 什么详细信息也不提供, 我也只能说感谢你的反馈.)  
(If you just say XXX doesn't work without providing any details, I can only say thank you for your feedback.)  
...

646185642, 2025-09-27:
大佬们

https://github.com/user-attachments/assets/fb62c1cc-54b0-4ef1-ab69-b2ceaabcd7f6

有解决方法吗？

wjz304, 2025-09-28:
xen 改成直接启动

646185642, 2025-09-28:
> 升级到v25.9.7，初步测试没有问题

大佬确定升级了没问题？

646185642, 2025-09-28:
> xen 改成直接启动

大佬，可以了。但是提示未检测到硬盘。需要怎么处理？/